breakintoprogram/agon-vdp #2: VDU 23,0 extensions

jgharston, 2022-10-31:
I can't find the code that processes the VDU stream, but from the code in BASIC I see you are using VDU 23,0 to pass various commands to the video processor.
This will clash with programs that already use VDU 23,0 for various things, for example it is common to see VDU 23,0,10,0,0,0,0,0,0,0 to get a solid block cursor, and VDU 23,0,10,32,0,0,0,0,0,0 to turn the cursor off. These would be transparently ignored if the platform doesn't implement it.

I would recommend using VDU 23,0,128+x for platform-specific extensions.

breakintoprogram, 2023-02-15:
Hi, thanks for the tip. I'll look into implementing for the next release.

breakintoprogram, 2023-03-21:
All operations in MOS and BBC BASIC now use defines rather than hard-coded constants

- mos_api.inc (for third party apps like BBC BASIC)
- defines.h (for MOS)

closed 2023-03-26 by breakintoprogram