breakintoprogram/agon-vdp #16: Increase quantity and quality of default modes in Agon

breakintoprogram, 2023-02-15:


breakintoprogram, 2023-03-05:
Example issues noted:

- https://www.facebook.com/groups/agoncomputer/posts/1173297166704997/

dsnopek, 2023-04-12:
Related to PR #37

julianregel, 2023-05-29:
I’ve been given this a lot of thought (and background reading!) as I can only get MODE 3 working properly on my Agon and VGA monitor.

I submit the following thoughts for consideration (sorry, long post!):

# BBC Modes

While the BBC modes with 256 vertical pixels have a nostalgic vibe, they don’t appear to be that compatible with most VGA monitors. Since the Agon isn’t trying to be a direct clone of the BBC, it would probably be best to ignore these and build on the VGA standard.

# VGA Modes

Of the modes that are included in the VGA standard, I’d suggest the following as being the most useful resolutions:

- 640x480 at 60Hz (current MODE 3) that maps to VGA Mode 12h
- 320x240 at 60Hz (proposed by dsnopek) that maps to QVGA/VGA Mode X

This would give a “high resolution” mode for displaying 80x25 text and graphics and a decent lower resolution with square pixels suitable for games. I think it would be reasonable to expect these modes to be compatible with pretty much all monitors.

It may also be useful to add:

- 320x200 at 70Hz that maps to VGA Mode 13h (albeit with reduced colours on the Agon) and is a resolution used by a lot of retro systems and games. As part of the VGA standard, this should also be supported by all monitors.

Logically, the other modes that could be generated from this combination of horizontal and vertical resolutions are:

- 640x200 at 70Hz as used by CGA and EGA
- 320x480 at 60Hz

I am not sure whether these last two options have real world uses beyond porting games and applications from other platforms that cannot be covered by the first three resolutions. I won’t reference these further.

# Logical Co-ordinates

The new resolutions would mean a change to the logical co-ordinates from 1280x1024 used by the BBC, but it appears that Acorn tackled this problem in RISC OS using its “OS Units” concept (https://www.riscosopen.org/wiki/documentation/show/Mode%20Variable%20Numbers). 

For example:

- RISC OS Mode 27: 640x480 = 1280x960 OS units
- RISC OS Mode 46: 640x200 = 1280x800 OS units

# Colours

The limited amount of memory in the ESP32 means that low colour modes are still useful. Suggested colour depths are 2, 4, 16 and 64 colours. 

# Proposed Modes

The following modes are suggested:

In terms of naming modes, the Acorn approach orders modes in decreasing resolution and then decreasing colour depth (leaving the "text" modes aside). While not essential to follow this pattern, it is a nice homage to the system that inspires it:

| Mode | Details |
|---|---|
| 0 | 640x480 in 16 colours at 60Hz |
| 1 | 320x240 in 64 colours at 60Hz |
| 2 | 320x200 in 64 colours at 70Hz |
| 3 | 640x480 in 4 colours at 60Hz |
| 4 | 320x240 in 16 colours at 60Hz |
| 5 | 320x200 in 16 colours at 70Hz |
| 6 | 640x480 in 2 colours at 60Hz |
| 7 | Reserved for Teletext mode currently under development | 
| 8 | 320x240 in 4 colours at 60Hz |
| 9 | 320x200 in 4 colours at 70Hz |
| 10 | 320x240 in 2 colours at 60Hz |
| 11 | 320x200 in 2 colours at 70Hz |

I did not include a 640x480 in 64 colours at 60Hz mode as this will require 300K for the frame buffer and I assume there isn’t sufficient memory in the ESP32 for this.

Hope this is helpful!

julianregel, 2023-07-02:
I have submitted PR [68 ](https://github.com/breakintoprogram/agon-vdp/pull/68) that implements the modes originally detailed above (plus other mode-related features).

breakintoprogram, 2023-08-09:
Merged PR #68

closed 2023-08-09 by breakintoprogram