breakintoprogram/agon-vdp #20: Allow the cursor blinking to be disabled

avalonbits, 2023-03-20:
Currently, if the cursor is on, it will always blink. This makes it using the cursor in a text editor annoying because it blinks even when you are going through lines.

Allowing it to not blink and to be always visible makes it a lot easier for follow char / line / page movements.

snakebyte69, 2023-04-10:
Something that changes the cusor type would be good. i.e.
No Cursor
Underline
Blinking Underline
Block
Blinking Block

stevesims, 2024-04-03:
the Console8 VDP 2.7.0 implements all of Acorn's cursor control VDU commands.  this will let you change the cursor to flash faster, be a steady (non-blinking) cursor, hide it, and change which character rows the cursor starts and ends on.  additionally Agon-specific enhancement commands have been added to allow the width of the cursor to be adjusted too

as well as that, full support for VDU 23,16,x,y has been implemented, which allows for further control of text cursor movement, allowing the horizontal and vertical directions to be flipped, and x and y axes to be swapped.