breakintoprogram/agon-vdp #21: Set VGA 8 colour mode pallete to match the first 8 colours of vga 16 colour mode.

avalonbits, 2023-03-21:
Currently the default pallete for 8 colour mode is rather boring (black, dark red, dark green, dark blue, red, green, blue, white).
We should set it to match the first 8 colours of the vga 16 colour mode pallete.

This needs to be done after a resolution  change because as part of chaning the resolution the default pallete is always set.

So, within `change_resolution`, after the call to setResolution, we can add:
```c
if (colours == 8) {
        VGAController->setPaletteItem(0, RGB888(0x00, 0x00, 0x00));
	VGAController->setPaletteItem(1, RGB888(0xAA, 0x00, 0x00));
	VGAController->setPaletteItem(2, RGB888(0x00, 0xAA, 0x00));
	VGAController->setPaletteItem(3, RGB888(0xAA, 0xAA, 0x00));
	VGAController->setPaletteItem(4, RGB888(0x00, 0x00, 0xAA));
	VGAController->setPaletteItem(5, RGB888(0xAA, 0x00, 0xAA));
	VGAController->setPaletteItem(6, RGB888(0x00, 0xAA, 0xAA));
	VGAController->setPaletteItem(7, RGB888(0xAA, 0xAA, 0xAA));
}
```

breakintoprogram, 2023-03-30:
Actions:

- Reset palettes after MODE (this needs to be done anyway)
- Make sure the palettes are sensible for all paletted modes (pull them from the colour lookup list)
  - 2: Black and White (Bright)
  - 4: Black, Red, Yellow, White (Bright)
  - 8: First 8 colours (Bright)
  - 16: First 16 colours (Half Bright/Bright)

breakintoprogram, 2023-03-31:
NB:

- The 4 and 8 colour paletted modes are not working correctly in FabGL.
- The 2 and 16 colour modes work fine.

lennart-benschop, 2023-04-23:
Based on 1.03 release version, I added some more modes: 800x600 4 colour and 640x512 8 colour and the palettes are not what they should be (4 and 8 colour paletted modes ware wrong). I did work around this in an earlier local version of the code, so I will check if I can find what I did then.

closed 2023-06-27 by breakintoprogram