breakintoprogram/agon-vdp #27: Issue when uploading version 1.03RC

warrengc3, 2023-03-24:
Hello,

I have an Agon Light 2 and when I try to upload the sketch for version 1.03rc I get the following error:

Arduino: 1.8.19 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 0, Core 1, None"

esptool.py v3.3
Serial port /dev/cu.usbserial-10
Connecting......Traceback (most recent call last):

  File "esptool.py", line 5387, in <module>
Chip is ESP32-PICO-D4 (revision 1)
  File "esptool.py", line 5380, in _main
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
  File "esptool.py", line 4802, in main
Crystal is 40MHz
MAC: 0c:8b:95:aa:16:cc
  File "esptool.py", line 1056, in flash_set_parameters
Uploading stub...
  File "esptool.py", line 495, in check_command
Running stub...
  File "esptool.py", line 468, in command
Stub running...
  File "esptool.py", line 413, in read
Changing baud rate to 921600
StopIteration
Changed.
[24080] Failed to execute script 'esptool' due to unhandled exception!
Configuring flash size...
An error occurred while uploading the sketch

Any help is apprecated.

warrengc3, 2023-03-24:
I was able to load this using a windows machine.  It must be a MACOS issue.

breakintoprogram, 2023-03-29:
Okay cool. Glad you got it working. Going to close this as this is probably best dealt with on the Arduino forums.

closed 2023-03-29 by breakintoprogram