breakintoprogram/agon-vdp #29: Add VDU commands for window operations

breakintoprogram, 2023-03-27:


lennart-benschop, 2023-05-21:
One thing: what should the mode parameters return when you call VDU 23, &86?
- the dimensions of the entire screen?
- the dimensions of the currently selected text and graphics windows
- the dimensions of the entire screen plus the full window parameters (origin + dimensions)? This would make the returned packet quite large.
Or should we have a separate VDU command to return the window parameters?

lennart-benschop, 2023-05-29:
VDU31 should position the cursor relative to the text window, so VDU31,0,0 should place it at the top left corner of that window. Currently it is positioned relative to the top left corner of the whole screen.

closed 2023-06-27 by breakintoprogram