breakintoprogram/agon-vdp #30: Reduce sprite flickering

breakintoprogram, 2023-03-28:


nihirash, 2023-04-04:
Maybe apply sprite changes only on vblank helps make it stabler? 

Like you collect sprite changes commands to list and apply it only on vblank.

HarmonicBrick, 2023-04-22:
Looks like Canvas::waitCompletion() should only be necessary before Canvas::getPixel() or similar methods. FabGL takes care of the rest. Like so:
```
diff --git a/video.ino b/video.ino
index 3845f84..f0af424 100644
--- a/video.ino
+++ b/video.ino
@@ -337,6 +375,7 @@ void sendScreenPixel(int x, int y) {
 	// Do some bounds checking first
 	//
 	if(p.X >= 0 && p.Y >= 0 && p.X < Canvas->getWidth() && p.Y < Canvas->getHeight()) {
+    Canvas->waitCompletion(false);
 		pixel = Canvas->getPixel(p.X, p.Y);
 		for(byte i = 0; i < VGAColourDepth; i++) {
 			if(colourLookup[palette[i]] == pixel) {
@@ -472,6 +511,7 @@ char get_screen_char(int px, int py) {
 
 	// Now scan the screen and get the 8 byte pixel representation in charData
 	//
+  Canvas->waitCompletion(false);
 	for(int y = 0; y < 8; y++) {
 		charRow = 0;
 		for(int x = 0; x < 8; x++) {
@@ -927,9 +967,9 @@ void vdu(byte c) {
 				Canvas->drawChar(charX, charY, ' ');
 				break;
 		}
-		if(doWaitCompletion) {
-			Canvas->waitCompletion(false);
-		}
 	}
 }
 
```

stevesims, 2023-11-01:
fab-gl's sprites work in an odd way. 

each sprite keeps a copy of the background underneath the sprite where it has been drawn.

whenever a sprite is moved, or something is drawn on the screen, _all_ of the sprites are erased from the screen, which is done by drawing their background, and then once the drawing operation has been complete re-drawing the sprite.

this, essentially, happens _all the time_.  any trivial drawing operation causes all the sprites to be erased and redrawn.  those sprites that are last in the list of sprites will get redrawn last.  the sprites that flicker when at the top of the screen are those furthest down the list of sprites.

a simple mitigation of this issue is to y-sort your sprites.

better solutions involve re-working big parts of how fab-gl does its sprite rendering.

breakintoprogram, 2023-11-11:
I'm going to mark this as 'Done' as we are now using a fork of FabGL (vdp-gl) - any issues with graphics primitives and sprites should be raised in that project.

closed 2023-11-11 by breakintoprogram