breakintoprogram/agon-vdp #33: VGAController->setResolution(SVGA_1024x768_60Hz) fails to return when Sony SDM-E96D Monitor Used as Display

oldpatientsea, 2023-04-05:
As discussed on Facebook, agon-vdp crashes on my setup (AgonLight2, Sony SDM-E96D) when the modeline passed to the function VGAController->setResolution(modeLine) is the one for Mode 0 defined by the FabGL macro "SVGA_1024x768_60Hz". 

The monitor powers off after warning there is no signal and the Agon needs to be reset before it responds again.

Serial debugging shows that the call to setResolution in change_resolution(int colours, char * modeLine) never returns. 

```
   if(modeLine) {                                   
        debug_log("do_setResolution: modeLine: (%s)\n\r", modeLine);
        VGAController->setResolution(modeLine);        // Set the resolution
        debug_log("end_setResolution: modeLine: (%s)\n\r", modeLine);
    }
```
Serial log output when switching first to mode 1 then to mode 0

```
vdu_mode: 1
do_setResolution: modeLine: ("512x384@60Hz" 32.5 512 524 592 672 384 385 388 403 -HSync -VSync DoubleScan)
end_setResolution: modeLine: ("512x384@60Hz" 32.5 512 524 592 672 384 385 388 403 -HSync -VSync DoubleScan)
do_modeChange: canvas(512,384), scale(2.500000,2.666667)
vdu_mode: 0
do_setResolution: modeLine: ("1024x768@60Hz" 65 1024 1048 1184 1344 768 771 777 806 -HSync -VSync)
```
[_At this point the monitor powers down. Agon remains unresponsive and serial debug output silent until device is reset._] 

Workaround: replace SVGA_1024x768_60Hz modeline for mode 0 with a new custom definition:

```
#define SVGA_640x512_60Hz "\"640x512@60Hz\" 54 640 664 720 844 512 513 515 533 -HSync -VSync DoubleScan"
```

oldpatientsea, 2023-04-07:
Update: this modeline kills the ESP32 even when the monitor is not connected. This is tested by running the following with the monitor disconnected.

```
   10 REM > Modetester
   20 REM Test if MODE switch kills display
   30 REM
   40 FOR M% = 3 TO 0 STEP-1
   50   MODE M% : PRINT "MODE ";M%
   60   SOUND 1, -10, 0, 5
   70   REM Wait for the monitor to catch up
   80   X = 0 : REPEAT : X=X+1 : UNTIL X > 10000
   90 NEXT M%
  100 GOTO 40
  ```
The problem therefore is with my particular Olimex board rather than the monitor.

oldpatientsea, 2023-04-09:
A second user on the facebook group has reported the same issue (name withheld for the moment as I have not asked their consent to share it). They have two different Olimex boards, one which manifests crashes with mode 0, the other does not. They have not noticed any obvious physical difference between the ESP32 chip used on both boards.

For reference, the following are the modelines that crash mine are:

The following modelines crash my ESP

/** Modeline for 640x382@60Hz resolution */ 
#define VGA_640x382_60Hz "\"640x382@60Hz\" 40 640 672 740 840 382 385 387 397 -HSync +VSync DoubleScan"

/** Modeline for 640x384@60Hz resolution */ 
#define VGA_640x384_60Hz "\"640x384@60Hz\" 40 640 672 740 840 384 384 386 397 -HSync +VSync DoubleScan"

/** Modeline for 800x300@60Hz resolution */ 
#define SVGA_800x300_60Hz "\"800x300@60Hz\" 40 800 840 968 1056 300 301 303 314 -HSync -VSync DoubleScan"

/** Modeline for 800x600@60Hz resolution */ 
#define SVGA_800x600_60Hz "\"800x600@60Hz\" 40 800 840 968 1056 600 601 605 628 -HSync -VSync"

/** Modeline for 1024x768@60Hz resolution */ 
#define SVGA_1024x768_60Hz "\"1024x768@60Hz\" 65 1024 1048 1184 1344 768 771 777 806 -HSync -VSync"

/** Modeline for 1024x768@70Hz resolution */ 
#define SVGA_1024x768_70Hz "\"1024x768@70Hz\" 75 1024 1048 1184 1328 768 771 777 806 -HSync -VSync"

/** Modeline for 1024x768@75Hz resolution */ 
#define SVGA_1024x768_75Hz "\"1024x768@75Hz\" 78.80 1024 1040 1136 1312 768 769 772 800 +HSync +VSync"

/** Modeline for 1280x600@60Hz resolution */ 
#define SVGA_1280x600_60Hz "\"1280x600@60Hz\" 61.5 1280 1336 1464 1648 600 601 604 622 -HSync -VSync"

/** Modeline for 1280x720@60Hz resolution */ 
#define SVGA_1280x720_60Hz "\"1280x720@60Hz\" 74.48 1280 1468 1604 1664 720 721 724 746 +hsync +vsync"

/** Modeline for 1280x720@60Hz resolution */ 
#define SVGA_1280x720_60HzAlt1 "\"1280x720@60HzAlt1\" 73.78 1280 1312 1592 1624 720 735 742 757"

/** Modeline for 1280x768@50Hz resolution */ 
#define SVGA_1280x768_50Hz "\"1280x768@50Hz\" 64.050004 1280 1312 1552 1584 768 784 791 807 -HSync -VSync"

====

tonedef71, 2023-04-21:
This is very interesting.  I have the Olimex AgonLight 2.  I cannot use MODE 2 (which is currently set to FabGL's VGA_320x200_75Hz modeline setting in the Quark VDP firmware) with my Sceptre X9 Gaming VGA monitor.  When MODE 2 is enabled, the monitor screen dims slightly and fills up quickly with all white pixels.  I almost expect smoke to come out of the monitor when that happens.

oldpatientsea, 2023-05-01:
@tonedef71 sounds alarming! This is probably a slightly different issue however with your monitor not supporting the resolution rather than your Agon not supporting it. Whatever is causing this issue actually causes the ESP32 to shut down/lock up whereas I assume that you can (quickly!) switch to a new mode and continue using the Agon.

tonedef71, 2023-05-01:
> @tonedef71 sounds alarming! This is probably a slightly different issue however with your monitor not supporting the resolution rather than your Agon not supporting it. Whatever is causing this [...]

It is possible that my Sceptre X9 Gamer monitor is damaged and should support this modeline (VGA_320x200_75Hz) but no longer does.   I have currently modified `MODE 2` in my build of the VDP to use modeline "VGA_320x200_70Hz" because 70Hz does work on my monitor for the 320x200 VGA resolution.

Modeline QVGA_320x240_60Hz also works on my monitor, so I gave this pull request a thumbs-up vote: https://github.com/breakintoprogram/agon-vdp/pull/37

breakintoprogram, 2023-08-13:
Hi, new modes have been added as per pull request #68 . Please can you let me know if this resolves the issues.

breakintoprogram, 2023-11-11:
Closing this card as I've had no more feedback in it since mid-August.

closed 2023-11-11 by breakintoprogram