breakintoprogram/agon-vdp #34: Allocate bitmaps in PSRAM over a certain size; BMP bitmap mode

HeathenUK, 2023-04-07:
Hi there,

In addition to the sound tinkering in my Pull Request I've been playing with code to stream off the shelf .bmp bitmap assets from the SD card to the VDP. It's lead me to two suggestions, the first of which I think is worthwhile regardless of the second:

**Bitmap allocation in memory**

At the moment you mandate bitmaps to be malloc'd in SRAM, which makes perfect sense if you're expecting them to be less than at most 64x64 for sprite work. But if you want to load in a one-off logo or similar you quickly run out of SRAM and enter undefined behavior territory - usually it just fails silently. PSRAM runs at around about 12MHz equivalent, so at the resolutions we're dealing with it's more than fast enough to stream bitmap data back to the canvas/framebuffer when demanded. I suggest the following change:

```
if (height*width > 4096) dataptr = (void *)heap_caps_malloc(sizeof(uint32_t)*width*height, MALLOC_CAP_8BIT); //PSRAM

else dataptr = (void *)heap_caps_malloc(sizeof(uint32_t)*width*height, MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL); // SRAM

bitmaps[current_bitmap].data = (uint8_t *)dataptr; //In either case, assign it.
```

I've even tested with _all_ bitmaps malloc'd to PSRAM and it doesn't seem to impact any of my existing code or the various sprite demos. But keeping smaller assets in SRAM as before seems a reasonable balance regardless, just in case.

**BMP mode**

As I mentioned, this came about because I was playing the best way to load in assets from SD on the ez80 side. We don't have the luxury of being able to ingest assets into RAM on the ez80 in order to play with them and stream them in the way the VDP expects for the existing examples. BMP files are the simplest off the shelf asset, and even they have two issues that are hard to get around when you're ingesting a file top-down one char at a time: pixel data is stored bottom-up, and in the form BGR.

My suggestion is a special variant of `cmd == 1` in the bitmap/sprite handling code, `cmd == 100`:

```
if(cmd == 100) {
//
// BMP mode. Read data to the new databuffer, then flip the whole map vertically (because BMP is stored bottom-up).
//
 	for(n = 0; n < width*height; n++) ((uint32_t *)bitmaps[current_bitmap].data)[n] = (uint32_t)0xFF000000 | ((uint32_t)readByte() << 16) | ((uint32_t)readByte() << 8) | ((uint32_t)readByte());
 	flip((uint32_t *)bitmaps[current_bitmap].data, width, height);
 	debug_log("vdu_sys_sprites: BMP bitmap %d - data received - width %d, height %d\n\r", current_bitmap, width, height);
}
```

This approach assumes the ez80 doesn't stream an alpha byte for the bitmap, since .bmp shouldn't contain one. It assumes 0xFF, but you could expand this code easily to toggle the alpha value based on the pixel's colour.

The flip() function in question is a dirt simple function to flip any framebuffer vertically, could be done without a function but is exactly the sort of thing we would struggle to do effectively on the ez80:

```
void flip(uint32_t* framebuffer, int width, int height) {
    uint32_t* row_buffer = (uint32_t*) malloc(sizeof(uint32_t) * width);
    int row_size = width * sizeof(uint32_t);

    for (int y = 0; y < height / 2; y++) {
        uint32_t* top_row = framebuffer + y * width;
        uint32_t* bottom_row = framebuffer + (height - y - 1) * width;

        memcpy(row_buffer, top_row, row_size);
        memcpy(top_row, bottom_row, row_size);
        memcpy(bottom_row, row_buffer, row_size);
    }

    free(row_buffer);
}

```

As with the sound PR, I recognise there's a balance here between keeping true to the Agon's design spirit with the ez80 firmly in control and the VDP providing the equivalent functions of a period-correct video or sound IC, and making it possible to fully utilise the hardware combination. The UART link helpfully acts as a downward pressure on going too crazy here, since it take ~10 seconds to transfer a 128x128 BMP file.

Regards,

Greg

HeathenUK, 2023-04-08:
As an addendum, this is where I finally landed in VDP for my "cmd 100":

```
uint8_t alpha = readByte();
debug_log("%u: Expecting a BMP of width %d and height %d with alpha %d", millis(), width, height, alpha);
if (alpha == 0) for (uint32_t n = 0; n < width*height; n++) ((uint32_t *)bitmaps[current_bitmap].data)[n] = (uint32_t)0xFF000000 | ((uint32_t)readByte() << 16) | ((uint32_t)readByte() << 8) | ((uint32_t)readByte());
else if (alpha == 1) for (uint32_t n = 0; n < width*height; n++) ((uint32_t *)bitmaps[current_bitmap].data)[n] = ((uint32_t)readByte() << 16) | ((uint32_t)readByte() << 8) | ((uint32_t)readByte()) | ((uint32_t)readByte() << 24);
flip((uint32_t *)bitmaps[current_bitmap].data, width, height);
```

This extra alpha/non-alpha byte mirrors the bitmap asset handling code on the ez80 side (my ez80 main.c can be seen in my fork for VDP), which now supports 32-bit BMPs with alpha channels, since we're forced to store the bitmaps as RGBA888 anyway. You can either use an 8/24 bit BMP and save 25% in transfer time, or have full alpha support - choices!

Greg

HeathenUK, 2023-04-28:
Now we have access to fread() and fseek() there are better ways to deal with the whole-array flip, so I don't suggest adding flip() any more, though the other lines stand.

HeathenUK, 2023-05-12:
Now we have access to a wider range of file handling functions in MOS I've been playing with doing the pixel reordering and buffer flipping in-memory on the ez80 for .BMPs that are small enough.

The impact of doing this onboard the ez80 rather than asking the faster VDP to handle the bit twiddling is on the order of about ~5%, so very small indeed and on that basis I think my second suggestion is much more of a "nice to have".

That said, I still think the first suggestion - to direct larger bitmaps to PSRAM - is worthwhile. It doesn't seem to impact performance at all and prevents you running out of ESP SRAM.

jonathan-teh, 2023-06-02:
Unless I've missed something, how about writing the BMP into the buffer directly, bottom-up, which saves flipping it later and memory for the extra row:
```c
for (int y = height - 1; y >= 0; --y) {
    for (int x = 0; x < width; ++x) {
        ((uint32_t *)bitmaps[current_bitmap].data)[y * width + x] = (uint32_t)0xFF000000 | ((uint32_t)readByte() << 16) | ((uint32_t)readByte() << 8) | ((uint32_t)readByte());
    }
}
```
Or retain the `readLong_b()` from the original and have the Z80 reorder the colours per-pixel, padding with alpha if necessary.
Though IMHO this should be a separate issue from the bitmap allocation, they're related but not dependent.

HeathenUK, 2023-06-03:
Yep fully agree, as I said there's essentially two issues here even if both are relevant to the wider task of handling larger off-the-shelf images.

It's really more of a philosophical question about how much you're comfortable offloading to [more] modern hardware. I've written code to parse and upload bitmaps to the VDP entirely using the original VDU 23 27 1 (no flip() used), with the RGBA twiddling and the bitmap reversing all done on the ez80. It works out to around 2/3rds as fast if you do it on the ez80 vs flipping and re-ordering it on the ESP32, which isn't terrible at all.

But in any case you *definitely* need the VDP to be able to deal with larger bitmaps!

stevesims, 2023-08-12:
I looked into memory allocation stuff quite a bit whilst I was implementing sample uploading in my PR #76 

If I understood [the documentation correctly](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html), just using plain `malloc` _should_ be fine, and a better choice than the current `heap_caps_malloc` call that's explicitly allocating to internal RAM.  With a plain `malloc` call, psRAM should get allocated as/when needed.

I opted to use `heap_caps_malloc(length, MALLOC_CAP_SPIRAM)` when assigning sample buffers to explicitly ensure samples got placed into psRAM, as they shouldn't need super-fast access.

stevesims, 2023-09-12:
Changes brought in #88 mean that all bitmap data is allocated into PSRAM - see https://github.com/breakintoprogram/agon-vdp/blob/128ff0887b1096d905dd43d03f37fa080646f3ad/video/vdu_sprites.h#L133

We can probably close this issue @breakintoprogram @HeathenUK

breakintoprogram, 2023-09-13:
Consider it done - thank you @stevesims

closed 2023-09-13 by breakintoprogram