breakintoprogram/agon-vdp #36: Incorporate Gerhard Skronn's Multi font into the VDP

snakebyte69, 2023-04-09:
Gerhard Skronn
https://github.com/anoriangit/agon-vdp

jackokring, 2023-11-02:
It would also be good to delete the first 32 characters of the teletext font, as they aren't visible, (control codes), and this would also match the default font which does not have them. It likely needs an `x-32` somewhere in the teletext code, ...

stevesims, 2024-10-14:
FYI the Console8 VDP variant (which runs on all Agon Light compatible machines) supports a new Font API from version 2.8.0.

that API uses buffers (and thus PSRAM) to store fonts, as per Lennart's suggestion above - indeed it supports just about everything suggested.  any number of fonts can be uploaded - the only limit is the 4mb of usable PSRAM.  variable width fonts are _not_ supported, but fonts of different sizes _are_ so fonts are no longer restricted to 8x8.  you can switch between fonts at any time, and print out text containing multiple fonts

the C8 VDP also provides ways to use the first 32 characters in any font, via the use of VDU 27 (just like Richard Russell's versions of BBC BASIC)