breakintoprogram/agon-vdp #39: eZ80->ESP32 UART connection fails if Olimex board booted without keyboard connected

oldpatientsea, 2023-04-13:
When booting the Olimex board without a keyboard connected, the MOS version string is displayed as garbled.

Seen in VDP 1.03 RC3 / MOS 1.03 RC3 and VDP 1.03 RC4 / MOS 1.03 RC4. No other versions tested yet for comparison.

Note this is untested on the original board and may not be specific to Olimex one.

Stepping through MOS code in debugger allowed MOS version string to be displayed clearly (suggesting it may be related to timing perhaps since this process introduces delays?).

Will add further information when I have had a chance to investigate.

breakintoprogram, 2023-04-13:
This is caused by the VDP taking longer to initialise if no keyboard is present due to the PS2Controller taking longer to complete - assume waiting for a keyboard response and not getting one. I've fixed this by moving the wait_eZ80 test before the keyboard initialisation. Works fine now. Good spot!

oldpatientsea, 2023-04-13:
Quick work! I've tested your change (based on your description since you haven't committed it yet) and can confirm it works.

[PASS] Agon boots with keyboard connnected
[PASS] Agon boots without keyboard connected
[PASS] If keyboard plugged in subsequent to Agon booting without one, it should work

>>  I've fixed this by moving the wait_eZ80 test before the keyboard initialisation.

It's your call but personally I think the experience is a little nicer if instead you move the keyboard initialisation _after_ the wait_eZ80 test.

```
	init_audio();
	copy_font();
  	set_mode(1);
	boot_screen();
	wait_eZ80();
	PS2Controller.begin(PS2Preset::KeyboardPort0, KbdMode::CreateVirtualKeysQueue);
	PS2Controller.keyboard()->setLayout(&fabgl::UKLayout);
	PS2Controller.keyboard()->setCodePage(fabgl::CodePages::get(1252));
	PS2Controller.keyboard()->setTypematicRateAndDelay(kbRepeatRate, kbRepeatDelay);
  ```
  
This way you see the VDP version line on the display appear immediately, then there is a pause, then the rest of the output appears.
  
It's clear to the user that it's MOS that is being a little slower to start up. It also means that if you flash a new VDP and restart the board with the jumpers still closed then you still get the VDP display line rather than a completely blank screen.

Arguably both make it easier to spot what is happening and are potentially less bewildering to the user.

Of course it also makes the "no keyboard" delay more salient and there may be side effects I don't know about if you do it in this order.
  
Just a suggestion!

closed 2023-06-27 by breakintoprogram