breakintoprogram/agon-vdp #40: Investigate slow text output

breakintoprogram, 2023-04-17:


breakintoprogram, 2023-04-17:
There was a wait_completion in the vdu function that was intended to wait for any graphical operations to complete. It was also waiting for text to be updated. This is probably over-zealous, so have modified the code to work as intended.

breakintoprogram, 2023-04-18:
Okay this seems to work - possibly add one final refinement:

- At the start of VDU, set a global variable (doWaitComplete) to false
- Any operation that requires a wait sets that to true
- At the end only do a waitComplete if that flag is set to true

nihirash, 2023-05-01:
Do we really need "doWaitCompletion = true" in bitmap blitting(draw bitmap) operation?

I didn't meet any issues with disabled this flag but got really big speed improvement or maybe my situation is kind of survivorship bias.

closed 2023-06-27 by breakintoprogram