breakintoprogram/agon-vdp #46: Add VDU4 and VDU5 support

julianregel, 2023-04-29:
Please could VDU4 and VDU5 support be added:

- VDU 4: Print at text cursor
- VDU 5: Print at graphics cursor

This would give the ability to switch the printing of text between the two cursors and allow for text positioning that is more precise than the default text positions (as set by the BASIC "TAB" instruction). For example:

```
10 MODE 1
20 VDU 4
30 PRINT TAB(4,4)"PRINTING AT TEXT CURSOR"
40 VDU 5
50 MOVE 640,512
60 PRINT "PRINTING AT GRAPHICS CURSOR"
70 VDU 4
80 PRINT "BACK AT TEXT CURSOR"
```

In the above example, the text at line 30 should be near the top of the screen, the text at line 60 roughly in the middle and the text at line 80 following on from the text at line 30.

breakintoprogram, 2023-05-19:
Should be a straightforward fix:

- Add a flag that is set on VDU 5, cleared with VDU 4, CLS, MODE.
- When printing a character, check the flag and use one of two print character routines
- When printing to graphic, just need to increment X and don't need to clear character square

breakintoprogram, 2023-05-21:
Note that printing character 0x7F (DEL) will do a correct backspace/delete, but the MOS editor does not use that character when deleting text on screen. So backspace when editing will not delete characters.

closed 2023-06-27 by breakintoprogram