breakintoprogram/agon-vdp #48: Enahncement: Add text window support (VDU 28).

lennart-benschop, 2023-05-07:
Now that I have implemented MODE7 in my own fork of the VDP, I see that some existing BBC Basic programs put control characters for foreground colour in the left column and then set a text window using VDU28, so these control codes will remain in place. It is also useful for interactively typing plotting commands, so the typed commands remain in a window separate from the graphics being plotted.

Therefore I think that VDU 28 is quite useful. VDU 28, left_column, bottom_row, right_column, top_row.
VDU 24 (graphics window) could also be implemented and of course VDU 26 to reset the windows to their defaults,which is full screen.
VDU24 probably requires some magic with Canvas->setClippingRectangle() to limit the extent of drawn lines to that window.
VDU28 can just be managed inside video.,ino, adding the required offsets and limit checks.

lennart-benschop, 2023-05-08:
One good reason to have this functionality is for having the scroll functionality (VDU 23,7) for just the text window. Being able to scroll just  a section of the screen is a killer feature we must have!

leventp, 2023-05-08:
Issue #29 already opened for this purpose.

breakintoprogram, 2023-05-19:
Closing as it is a duplicate of #29

closed 2023-05-19 by breakintoprogram