breakintoprogram/agon-vdp #49: Implement the mode parameter of GCOL for drawing commands

breakintoprogram, 2023-05-16:


breakintoprogram, 2023-05-17:
Think the only modes FabGL can support are: 0, 4, 6. All other modes will have no effect as per 5. 

Not sure how the swap fg/bg colour option works.

lennart-benschop, 2023-05-17:
Swap fg/bg is supposed to work like this: for every point you visit:
- if it has the foreground colour, change it to background colour.
- else if it has the background colour, change it to foreground colour.
- else leave the point unchanged.

lennart-benschop, 2023-05-17:
I would also implement the distinct 0..7 plot modes:
0..3 use relative coordinates.
4..7 use absolute coordinates
0,4 move, do not draw anything.
1,5 draw in foreground colour.
2,6 draw in inverse colour (this is FabGL's NOT operation I think).
3,7 draw in background colour
 FabGL appears to have inverse colour support only for dots and lines, but it would be good to implement all these modes for them. Relative coordinates, fg and bg drawing can also be implemented for the other graphics primitives.

breakintoprogram, 2023-05-17:
That's not part of the scope for this card.

lennart-benschop, 2023-05-18:
I'll make another card for it then.

closed 2023-06-27 by breakintoprogram