breakintoprogram/agon-vdp #51: Implement relative plotting modes and background/inverse.

lennart-benschop, 2023-05-18:
I would like to implement the distinct 0..7 plot modes:
0..3 use relative coordinates.
4..7 use absolute coordinates
0,4 move, do not draw anything.
1,5 draw in foreground colour.
2,6 draw in inverse colour (this is FabGL's NOT operation I think).
3,7 draw in background colour
FabGL appears to have inverse colour support only for dots and lines, but it would be good to implement all these modes for them. Relative coordinates, fg and bg drawing can also be implemented for the other graphics primitives.

breakintoprogram, 2023-08-03:
Check the comments in #74 to ensure that everything is covered.

stevesims, 2023-09-12:
I've implemented as much of this as is practical given the limitations of fab-gl.  PR on the Console8 codebase is here: https://github.com/AgonConsole8/agon-vdp/pull/43

I'll make a PR for this codebase in the morning.

stevesims, 2023-09-26:
further to this - the PR I made got written, and has since been merged.  (see #89 )

support for the "inverse" plot operations specifically was _not_ included, since fab-gl doesn't fully support that functionality.  so any calls that attempt to use "inverse" operations get ignored.

as it goes, I misunderstood the limitations of fab-gl's `NOT` operation, which is it's limited "inverse" support - I'd thought it would only be applied to horizontal and vertical line drawing, but it's actually any straight line.  that's not _quite_ as restricted as I thought, so it _is_ more useful than I'd originally thought and may be worth including.

I've got an experimental PR at AgonConsole8/agon-vdp#55 that enables inverse for straight line plotting calls.  I'm not entirely happy with it tho - there's a fallback in there for non-straight-line calls to draw in "foreground" colour, and right now I'm not entirely sure what will happen with "fill" operations.

breakintoprogram, 2023-09-27:
I'm going to close this card now given the recent updates. Any further refinements / bug fixes can be dealt with in new cards.

closed 2023-09-27 by breakintoprogram