breakintoprogram/agon-vdp #58: <CR><LF> doesn't update to the new background colour, it needs to print actual text to update.

sijnstra, 2023-06-06:
Code to replicate the error:
10 CLS
20 FOR I = 1 TO 6
30 VDU 17, 129
40 PRINT "RED BACKGROUND";
50 VDU 17, 128
60 FOR J = 1 TO 6 : PRINT CHR$(13);CHR$(10);: NEXT
70 PRINT "BLACK BACKGROUND";
80 FOR J = 1 TO 6 : PRINT CHR$(13);CHR$(10);: NEXT
90 NEXT
Produces the following:
[image]

oldpatientsea, 2023-06-07:
Is this related to the same underlying issue discussed in  VDU 23,16 support for controlling cursor movement and screen scrolling #55  ?

sijnstra, 2023-06-08:
Unsure? I see #55 is an enhancement, and yes this may be related, however, I am reasonably certain this issue existed in 1.03 although I haven't re-flashed to be sure.

sijnstra, 2023-06-09:
> Is this related to the same underlying issue discussed in VDU 23,16 support for controlling cursor movement and screen scrolling #55 ?

I've had a deeper look, at the posts around 55, and it seems that #55 has another feature I would love to work as described, but doesn't seem to behave this way yet. I do not believe these features are related but happy to be corrected if I miss something. The feature I would love to see is:
<html>
<body>

b0=1 | enables scroll protection, the text cursor moving off the positive edge of the window generates a pending newline, and waits at the edge of the window until the movement is needed. While the cursor is waiting, the cursor position returned by POS via OSBYTE &87 is the maximum column position plus one, ie the window width (eg 80 in MODE 0). If the cursor is moved elsewhere, the pending newline is abandoned, and no scrolling occurs.
-- | --

</body>
</html>b0=1	enables scroll protection, the text cursor moving off the positive edge of the window generates a pending newline, and waits at the edge of the window until the movement is needed. While the cursor is waiting, the cursor position returned by POS via [OSBYTE &87](https://beebwiki.mdfs.net/OSBYTE_%2687) is the maximum column position plus one, ie the window width (eg 80 in MODE 0). If the cursor is moved elsewhere, the pending newline is abandoned, and no scrolling occurs.

sijnstra, 2023-09-20:
Not sure what's causing this, but I can't replicate in Terminal mode.

stevesims, 2024-01-15:
the underlying cause for this issue was that the fab-gl canvas fills in the newly scrolled-in background with the currently selected graphics foreground colour, if one has been set.

Acorn systems would fill the scrolled-in area with the current text background colour.

we were not explicitly setting any colour at all on the canvas.  as a result under some circumstances you'd see a filled-in background being used because a drawing operation had been completed, but often/usually you'd just get a black background.

this is now fixed in #159

stevesims, 2024-10-14:
further to my earlier comment, the implementation of "scroll protection" cursor behaviour in the Quark VDP differs significantly from the system Acorn described and implemented.  what was implemented was, essentially, a form of scroll prevention

the Console8 VDP has full support for all of the various cursor behaviours defined for VDU 23,16 which includes proper scroll protection.  other VDU commands related to controlling cursor appearance and function have also been implemented.  these have been available since VDP 2.7.x with some fixes and enhancements made in the 2.8.x releases