breakintoprogram/agon-vdp #66: sprites_4.bas causes VDP to shut down after several minutes

jonathan-teh, 2023-06-27:
I suspect this is a VDP bug but since the reproducer is in BBC BASIC I've raised it here. Straightforward to reproduce:
```
LOAD "SPRITES_4.BAS"
RUN
```
For me, after between 1 min 30 s and 5 minutes (across 4 runs), VDP appears to shut down, the monitor no longer detects a signal and shuts off, the keyboard is unresponsive to any input e.g. to toggle caps lock or num lock. Pressing the reset button is the only way back to a usable machine.
`sprites_5.bas` can also cause a shutdown but takes longer, 12 minutes on the first run, the second run lasted 16 minutes before I stopped it.
MOS 1.03, VDP 1.03, BBC BASIC 1.04 on an Agon Light 2. Has been reproduced on another Agon Light 2 running the same software versions.

breakintoprogram, 2023-11-15:
This seems to be related to animating the sprite (VDU 23, 27, 8). I've removed that from the demo and all is well.

breakintoprogram, 2023-11-15:
In sprites.h, changing the function nextSpriteFrame to this seems to fix the issue

```
void nextSpriteFrame() {
	auto sprite = getSprite();
	auto frame = sprite->currentFrame;
	sprite->setFrame(frame < sprite->framesCount -1 ? frame + 1 : 0);
}
```
which seems to indicate an issue with `sprite->nextFrame()`

breakintoprogram, 2023-11-15:
I think I see what is happening here:
This could could for a brief period set currentFrame to an invalid frame before the if condition is triggered.
If a sprite were to be drawn during that period, then you'd get a guru exception.
```
void nextFrame() { ++currentFrame; if (currentFrame >= framesCount) currentFrame = 0; }
```
I've modified the code locally in vdp-gl as follows (~line 571 of displayController.h) and am testing now.
```
void nextFrame() { currentFrame = (currentFrame  < framesCount - 1) ? currentFrame + 1 : 0; }
```

breakintoprogram, 2023-11-15:
This has now been running for over 90 minutes on my test Agon, whereas before it would last no longer than 5-10 minutes. I've issued a pull request on the vdp-gl repo here https://github.com/avalonbits/vdp-gl/pull/31.
As this is not a bug in agon-vdp, and there will be no changes on my side, I'm going to close this card as 'Won't Do'.

closed 2023-11-15 by breakintoprogram

jonathan-teh, 2023-11-15:
Thanks for fixing this. Should the fix also be sent upstream to FabGL?

stevesims, 2023-11-16:
excellent investigation work @breakintoprogram 

I'd never managed to reproduce this error when I'd tried - left sprites_4 running for many many minutes, maybe even hours before with it seemingly quite happy.  the unlikely circumstance of the `nextFrame` method getting interrupted between its two statements would make it an incredibly rare crash indeed.  :grin:

I've been trying to fix sprites performance to reduce flickering for the past couple of weeks, and have been incidentally trying to look out for this bug.  I _have_ seen occasional/rare crashes when testing out my revised sprites code; i had put that down to bugs in my own code, which I couldn't see, but at least some of the crashes was probably this bug :grin:

breakintoprogram, 2023-11-16:
Cool, thanks @stevesims. It's a quick fix. Ideally it should fail gracefully if an invalid sprite frame is selected.