breakintoprogram/agon-vdp #67: mode 0 is the problematic mode for me

ukscone, 2023-07-01:
I keep an eye on progress/changes in the mode definitions as i've always at least 1 problem mode in the vdp in all vdp versions so tonight i saw that mode 2 definition had been changed and that now seems ok on all my monitors, a 13" no name cheapo amazon thing, a 24" amazon fire tv and a 20" Sceptre but mode 0 is still a right pita. either says unsupported on the 13" that's max res is 1366x768@60Hz, and just weird on the sceptre that can do upto 1600x900@75Hz, offset with a shakey, "buzzy" green vertical line about 20 pixels to the left of the text. i can move it off screen by messing with the monitor settings but it doesn't stick (that's monitor fault but hey ho) and the fire tv is also weird and iffy but each time the way it exhibits it's iffyness it's slightly different. is it possible to try to find a "more supported" standard high resolution for 2 colour mode 0 please as i can't be the only person where mode 0 is a pita if all 3 of my possible monitors are iffy with the current resolution

jonathan-teh, 2023-07-10:
Having tested mode 0 on various displays, I've found 2 (fairly old) LCD TVs that do not work: Dell W1900 and Sony KDL40V3000, showing a black screen with "unsupported video mode" or words to that effect.

It works on several monitors but the OSDs report a vertical refresh rate of 57-58Hz instead of 60Hz. Specifically:

| Model | Native | OSD |
| --- | --- | --- |
| hp L1530 | 1024x768 | 1024x768 57Hz |
| hp LP2065 | 1600x1200 | 1280x800 48.4kHz 58Hz |
| Dell ST2310 | 1920x1080 | 1024x768 58Hz |
| Philips 107B | CRT (max 1920x1440) | new mode 47kHz 58Hz |

All the displays work fine with the double-scanned 512x384 60Hz mode 1, reporting 1024x768 60Hz in the OSD.

The modeline in FabGL:
```
#define SVGA_1024x768_60Hz "\"1024x768@60Hz\" 65 1024 1048 1184 1344 768 771 777 806 -HSync -VSync"
```
is identical to the one X.org on Linux uses:

```
Modeline "1024x768"x60.0   65.00  1024 1048 1184 1344  768 771 777 806 -hsync -vsync (48.4 kHz e)
```
which could suggest that FabGL's timing is slightly slow, perhaps by 4-5%, and that's enough for the TVs to fail to sync.

As an aside, with the currently problematic (until 1.04 is released) 320x200 75Hz mode 2, all the monitors do work but report a vertical refresh rate of 69Hz, 8% slower than what it's supposed to be. All claim to support a vertical refresh rate of at least 76Hz.

I don't have an oscilloscope so I can't verify the accuracy of the OSDs.

breakintoprogram, 2023-08-13:
Hi, new modes have been implemented as per pull request #68. Please can you let me know if this resolves the issues.

breakintoprogram, 2023-11-11:
I think all the mode issues have been resolved now. Going to tentatively close this as 'Done'

closed 2023-11-11 by breakintoprogram