breakintoprogram/agon-vdp #70: Compile error with VDP 1.04 RC1 source code

leventp, 2023-07-11:
When I am compiling the latest VDP 1.04 RC1 source code (provided with the release), I am getting the error below.  

:\Users\lpanc\Downloads\agon-vdp-1.04-RC1 (1)\agon-vdp-1.04-RC1\video\video.ino: In function 'void updateRGB2PaletteLUT()':
video:604:47: error: 'void fabgl::VGAPalettedController::updateRGB2PaletteLUT()' is protected within this context
   case 2: VGAController2.updateRGB2PaletteLUT(); break;
                                               ^
In file included from C:\Users\lpanc\Documents\Arduino\libraries\FabGL\src/dispdrivers/vga2controller.h:51,
                 from C:\Users\lpanc\Documents\Arduino\libraries\FabGL\src/fabgl.h:319,
                 from C:\Users\lpanc\Downloads\agon-vdp-1.04-RC1 (1)\agon-vdp-1.04-RC1\video\video.ino:43:
C:\Users\lpanc\Documents\Arduino\libraries\FabGL\src/dispdrivers/vgapalettedcontroller.h:117:8: note: declared protected here
   void updateRGB2PaletteLUT();
        ^~~~~~~~~~~~~~~~~~~~
video:605:47: error: 'void fabgl::VGAPalettedController::updateRGB2PaletteLUT()' is protected within this context
   case 4: VGAController4.updateRGB2PaletteLUT(); break;
                                               ^
In file included from C:\Users\lpanc\Documents\Arduino\libraries\FabGL\src/dispdrivers/vga2controller.h:51,
                 from C:\Users\lpanc\Documents\Arduino\libraries\FabGL\src/fabgl.h:319,
                 from C:\Users\lpanc\Downloads\agon-vdp-1.04-RC1 (1)\agon-vdp-1.04-RC1\video\video.ino:43:
C:\Users\lpanc\Documents\Arduino\libraries\FabGL\src/dispdrivers/vgapalettedcontroller.h:117:8: note: declared protected here
   void updateRGB2PaletteLUT();
        ^~~~~~~~~~~~~~~~~~~~
video:606:47: error: 'void fabgl::VGAPalettedController::updateRGB2PaletteLUT()' is protected within this context
   case 8: VGAController8.updateRGB2PaletteLUT(); break;
                                               ^
In file included from C:\Users\lpanc\Documents\Arduino\libraries\FabGL\src/dispdrivers/vga2controller.h:51,
                 from C:\Users\lpanc\Documents\Arduino\libraries\FabGL\src/fabgl.h:319,
                 from C:\Users\lpanc\Downloads\agon-vdp-1.04-RC1 (1)\agon-vdp-1.04-RC1\video\video.ino:43:
C:\Users\lpanc\Documents\Arduino\libraries\FabGL\src/dispdrivers/vgapalettedcontroller.h:117:8: note: declared protected here
   void updateRGB2PaletteLUT();
        ^~~~~~~~~~~~~~~~~~~~
video:607:49: error: 'void fabgl::VGAPalettedController::updateRGB2PaletteLUT()' is protected within this context
   case 16: VGAController16.updateRGB2PaletteLUT(); break;
                                                 ^
In file included from C:\Users\lpanc\Documents\Arduino\libraries\FabGL\src/dispdrivers/vga2controller.h:51,
                 from C:\Users\lpanc\Documents\Arduino\libraries\FabGL\src/fabgl.h:319,
                 from C:\Users\lpanc\Downloads\agon-vdp-1.04-RC1 (1)\agon-vdp-1.04-RC1\video\video.ino:43:
C:\Users\lpanc\Documents\Arduino\libraries\FabGL\src/dispdrivers/vgapalettedcontroller.h:117:8: note: declared protected here
   void updateRGB2PaletteLUT();
        ^~~~~~~~~~~~~~~~~~~~
exit status 1
'void fabgl::VGAPalettedController::updateRGB2PaletteLUT()' is protected within this context

tonedef71, 2023-07-11:
> When I am compiling the latest VDP 1.04 RC1 source code (provided with the release), I am getting the error below.
> 

I think you simply need to uninstall the fab-gl library (1.0.8) and then install the vdp-gl (1.0.3).  That should make those compilation errors go away.

leventp, 2023-07-11:
That worked. Thanks.

closed 2023-07-11 by leventp