breakintoprogram/agon-vdp #71: VDU 24 behaves differently from original BBC BASIC

leventp, 2023-07-12:
Looks like when setting a graphic  window using VDU 24, the origin is automatically set to  the bottom left coordinate. in Original BBC BASIC it is set with an additional VDU 29 statement. Used the code below to test this behavior:

10 MODE 1
20 VDU 24,150;300;1100;700;
30 GCOL 0,129: CLG
40 MOVE 0,0
50 DRAW 1100,700

To have the same effect on my BBC Micro, I had to add  the following line:
35 VDU 29,150;300;

Tested with the latest releases.

breakintoprogram, 2023-08-03:
Ok, should be an easy fix.

stevesims, 2023-11-01:
Verified.

Took a little while to work this one out, as at first glance it looks like all the graphics code is doing the right thing.

Long story short, when you've set a graphics viewport, calls within the VDP code for working out graphics point positions (in the `translateViewport` call) were getting their pixel locations adjusted to be relative to the viewport as well as the origin.  they shouldn't.  that routine should _only_ be doing its calculations relative to origin.

I'll raise a PR in the morning.

stevesims, 2023-11-02:
PRs raised with a fix

long story short, removing viewport adjustment from `translateViewport` turns that function into a clone of `translateCanvas`, so it's been got rid of

leventp, 2023-11-02:
Thanks Steve. Also appreciate refactoring of the code.

PR #131 (2023-11-06): fix VDU 24
VDU 24 sets the graphics viewport.  essentially it sets a clipping rectangle, and that should be all it does…

this fixes a bug whereby setting the graphics viewport would also effectively change the graphics origin.  this is contrary to how Acorn handles VDU 24.  (additionally setting the origin using VDU 29 would make for a double-offset)

all uses of `translateViewport` have been changed to use `translateCanvas` instead.  this should make behaviour more understandable, and more predictable, and consistent with Acorn.

besides fixing the observed VDU 24 issue, this PR changes the  [... cut]
closed 2023-11-06 by breakintoprogram