breakintoprogram/agon-vdp #72: VDU 14 (Paged mode) does not clear on screen mode change

julianregel, 2023-07-15:
If VDU 14 is set, the screen pauses at the point of scrolling until Shift is pressed.

The expected behaviour (as seen on the BBC) is that a mode change unsets this and turns paged mode off.

This does not happen on the Agon, which causes unexpected behaviour when running programs that change mode and assume the page scroll state is off.

breakintoprogram, 2023-08-03:
Okay, should be an easy fix.

stevesims, 2023-09-12:
I've noticed that paged mode still has some behaviour differences when compared to the BBC Micro.  I made some notes about it here: https://github.com/AgonConsole8/agon-vdp/issues/39

There's all sorts of things about what Acorn did that are kinda subtle, and make for a better user experience.  As I've now worked thru what that behaviour is hopefully I'll be able to get around to trying to implement it.

closed 2023-11-11 by breakintoprogram