breakintoprogram/agon-vdp #73: VDP Documentation Errors/Bugs

julianregel, 2023-07-15:
The VDP documentation at https://github.com/breakintoprogram/agon-docs/wiki/VDP has a few errors/bugs:

# VPD vs VDP

There are multiple instances of VPD instead of VDP (needs a global search and replace to fix).

# Incorrect VDU statement for bitmap functionality

The description of VDU 23,27 states:

```
VDU 23. 27, 1, n, w; h; b1, b2 ... bn: Load colour bitmap data into current bitmap
VDU 23, 27, 2, n, w; h; col1; col2; b1, b2 ... bn: Load monochrome bitmap data into current bitmap
```

The "n" parameter is not required and if used, causes the command to fail (this caught me out for a while!).

jonathan-teh, 2023-07-18:
Also, the second one should read (from reading the code):
```
VDU 23, 27, 2, col: Set current bitmap to a solid colour
```
The original description gives the impression that it's one way of greatly reducing the bandwidth requirements (1/32 of the colour option) for certain applications. Are there any plans to implement monochrome bitmaps? Or colour RGBA2222 bitmaps (1/4 the bandwidth)?

radekh, 2023-07-21:
`VDU 23,27,1,w;h;b1,b2,...bn   -- Load colour bitmap`
Be aware that b1, .... are 32bit RGBA.   so pushing by components, n is w*h*4.

`VDU 23,27,2,w;h;col1;col2   -- Create solid one colour bitmap`
Be aware that the colour is 32bit RGBA again.  so push two sixteen bit or four eight bit values there.

Yes the n parameter in these was/is typo.

I got the knowledge there from looking inside of VDP source code.  I had an issue with "documentation" so I start looking through the source version 1.03 and 1.04RC1.

breakintoprogram, 2023-08-03:
I've fixed some of the documentation but as stated it's still not entirely clear. Will update.

closed 2023-11-11 by breakintoprogram