breakintoprogram/agon-vdp #79: Moving a blinking text cursor may force it to not blink anymore (in mode 1 at least in  1.04rc1 mos/vdp)

theflynn49, 2023-08-18:
The title may be misleading. I am using asm through z88dk C compiler.
What happens is that the blinking of the text cursor is not visible unless the current foreground and background colors are the same than the colors under the cursor.
So, moving the cursor in a text area with mixed colors gets very complicated : you need to
- move the cursor (vdu 31 x y)
- fetch the colors in that position if not known
- set colors to that position's colors (vdu 17 x; vdu 17 x+0x80) but it's not enough
- fetch the current character in that position if not known
- rewrite the character to 'validate' the colors somehow
Only then you are sure that the cursor will blink.
If you play with the colors, you may obtain a situation where only the displayed character blinks, not the complete font space.
 The emulator (1.03/1.03) doesn't show this problem

So maybe that was intended, but somehow I think this is much more complicated than it should be.
 
How to reproduce quickly : under MOS type : 
*vdu 31 0 22 17 15 17 128 65 65 17 14 17 130 66 31 0 22 <enter>
You should see "*AB" and the A should blink but it doesn't
change the color 14 by "0" and you'll see it blink
*vdu 31 0 22 17 15 17 128 65 65 17 0 17 130 66 31 0 22 <enter>
change the same parameter to "15" and you'll see that only the character "A" blinks
*vdu 31 0 22 17 15 17 128 65 65 17 15 17 130 66 31 0 22 <enter>

Rgds ...

breakintoprogram, 2023-09-04:
Hi @theflynn49 thank you for your comment. I've never been that happy with the cursor implementation and will probably get around to rewriting it at some point. At the moment it currently inverts a rectangle at the text cursor position.

stevesims, 2024-04-03:
the problem with the cursor drawing essentially is that it swaps the current text foreground and background colours within the cursor rectangle.  given the capabilities of vdp-gl this was a sensible compromise, but it does means that if the cursor is sitting above a rectangle of the screen that doesn't contain the current text colours the cursor will be invisible.

a solution to this is to instead draw the cursor using an XOR plotting system.

the difficulty with this idea is that the most recently published version of vdp-gl does not support XOR plotting.  I have however implemented that functionality, and it has been merged into the codebase - it just hasn't been released yet.

the latest Console8 VDP release (2.7.1) uses an XOR-based plotting mechanism for the cursor (it is built using PlatformIO so does not need to use the published version of vdp-gl and can instead use a custom branch containing this functionality).  it is drawn using XOR with _both_ the current foreground text colour and current background text colour, which ensures that the cursor will remain visible when the cursor is active and set to be visible.  this can mean that the cursor may appear in slightly unexpected colours when placed on a background that doesn't match the current text colours, but I felt that was a reasonable compromise

theflynn49, 2024-04-07:
Sounds good to me. It's been a while, but what I remember is that to get a reasonable result, I had to redraw the characters below the cursor when it moved, and methink this is bad : you should be able to move the cursor without knowing what's underneath ...