breakintoprogram/agon-vdp #81: Monochrome bitmap VDU command doesn't match documentation

stevesims, 2023-08-28:
The [documentation for the bitmap commands](https://github.com/breakintoprogram/agon-docs/wiki/VDP#bitmaps) says that `VDU 23, 27, 2, ...` should be for uploading a monochrome sprite to the VDP, and accept width, height, two colours and data.

[The implementation](https://github.com/breakintoprogram/agon-vdp/blob/4045c7b8285fc6fcdf88fad2a0cd6ed8e2527200/video/video.ino#L1797C8-L1797C8) however does not accept any data and accepts only a single colour.  It just creates a solidly filled bitmap.

The version of the command in the documentation is probably more useful.... I can't really imagine much of a use-case for the current implementation.

As an alternative, we may wish to consider supporting other bit depths of bitmaps than the current `PixelFormat::RGBA8888` only system.  We could re-purpose `VDU 23, 27, 2` for extended/enhanced bitmap uploading.

breakintoprogram, 2023-09-04:
Hi @stevesims yes well spotted, documentation error. I'll consider your change request, though as it is a breaking change I will probably not use VDU 23, 27, 2 in case it is being used in the wild.

stevesims, 2023-09-04:
my feeling is that, at this time, there's not all that much use of any particular features in the wild - the community is still pretty small, so we can probably break a few things.  indeed the revised screen mode numbers is already a breaking change, one that's pretty easy for developers to fix their code

in the case of the current `VDU 23,27,2` as I said I can't really imagine much of a use-case.  creating a solid bitmap of a single colour when there's no mechanism for adjusting the contents of that bitmap doesn't deliver _much_.  if the bitmap is being plotted to the screen a developer is more likely to have drawn a filled rectangle for the same functionality.  that leaves using a solid colour bitmap for a sprite frame, and that doesn't feel very useful either.  much like the change in screen modes, these are fairly easy things to work around.

then again, there's plenty more command bytes that can be used for `VDU 23,27` - the only real reason to re-purpose 2 would be to keep bitmap command numbers together, separate from the sprite commands

jonathan-teh, 2023-09-07:
I suggested an update to the documentation [here](https://github.com/breakintoprogram/agon-vdp/issues/73#issuecomment-1641088693) to match what it actually does.

closed 2023-11-11 by breakintoprogram