breakintoprogram/agon-vdp #84: Keyboard layout to be fixed, SD card recognition after reset

ZbigniewB, 2023-08-31:
Hello, using current 1.03 versions of MOS and VDP and I noticed a few flaws:
— presently the keyboard has quote (") and at (@) characters mutually swapped. Could be possible to fix this? It's rather irritating
— no tilde (~) character present, which is replaced by space
— SD card (mass storage) is properly recognized after it is inserted only when MOS is already running. But directly after reset Agon always complains "no valid FAT filesystem". Could be possible to enforce SD card detection (and mounting, if present) after each reset?

And a question: when doing VDP upgrade 1.02->1.03 in Arduino IDE, at the end of programming it said "going to enforce RTS reset" (or something like this) — but in the end it reported nothing more, like it was unable to do that reset by itself (I waited quite long). Is it normal?

stevesims, 2023-09-01:
Before I address your questions, you should note that there has been several updates both to MOS and the VDP that change behaviour a bit, so you may wish to update.

On your first query, [you can change the keyboard layout](https://github.com/breakintoprogram/agon-docs/wiki/MOS#keyboard-layout).  The default is for a British keyboard.  If you have a US keyboard from the MOS command prompt you'll need to type `SET KEYBOARD 1`.  NB this command is currently case sensitive.  Many people have an `AUTOEXEC.TXT` file on the root of their SD card that includes such a command.

The tilde character works for me when I have the correct keyboard set.

Currently the SD card can't be ejected or re-mounted.  There is a pull request for MOS that adds a `MOUNT` commands, as well as some other useful additions.  Until that has been merged, the only way to re-mount an SD card is to press the reset button.

ZbigniewB, 2023-09-01:
Oh, so… US keyboard is in much more common use, then I assumed it is the default and the swapped characters are a flaw. OK, thanks — at the moment I'll save myself with that autoexec tip.
I noticed one more flaw regarding SD cards: Agon is unable to use older, small SD cards of 1 GB size (which would be more than enough for 8-bit computer). Whatever I will set (tried every possible combination) as FAT-type and cluster size — Agon all the time complains „no valid FAT filesystem” (4 GB and 8 GB are „seen” OK).
I'm aware these smaller cards are slowly coming out of use, still it may mean some flaw in filesystem implementation, that may have also other negative consequences.

breakintoprogram, 2023-09-04:
Hi @ZbigniewB thank you for your comments.

- As suggested, use the SET KEYBOARD statement in the autoexec.txt to set the default keyboard on start
- The SD card mounting is done on reset automatically, but not when a card is inserted during normal operation
- The VDP reset should be automatic after upload. This is a function of the Arduino IDE / ESP32 tools, not the VDP software
- Support for older cards is limited due to their class / speed a class 10 is recommended, though slower ones may work

If you are happy with these answers I'll mark this card as 'Won't Do' (no software mods required).

ZbigniewB, 2023-09-04:
Well OK — if you think it's not related to any bug, I'm fine. Thanks for your prompt reply!

closed 2023-09-15 by breakintoprogram