breakintoprogram/agon-vdp #101: Feature Request: ESP32 code access to eZ80 connections, e.g. SDCard

snakebyte69, 2023-09-29:
Can a simple routine be added to MOS that would allow FabGL, ESP32Tiny apps access to the SDCard and any other hardware that requires the eZ80.

The existing ESP32 apps work for the most part but are limited because MOS has no ability to process requests.

breakintoprogram, 2023-11-16:
Yes it is possible, though the performance may not be brilliant:

- MOS will have to send a packet requesting the GPIO from the ESP32
- VDP will then have to read the data and send the results back to MOS
- MOS will then need to process that packet

snakebyte69, 2023-11-17:
Performance would not be critical at first just getting it working would be a proof of concept and would mean all the FabGL and ESP32Tiny apps would work.
e.g. there is existing MAME code that could load ROMs from the SDCard.
Games will bring a LOT more people to the platform.

Also in this mode MOS would only be running commands from the ESP32.

stevesims, 2023-11-17:
so let me preface my comment by saying that I do understand the desire for this...

the idea of adding features to agon-vdp in order to support running FabGL or ESP32Tiny apps presumes the ability to run those apps within the current VDP software platform.  agon-vdp simply is not architected to allow that to be possible.

FabGL and ESP32Tiny apps are written to run on the metal.  they are inherently something _other_ than agon-vdp, and not really things that can be made to run "inside" agon-vdp

as I see it, this is therefore not a feature that belongs as part of the Agon software platform.  software written for the Agon software platform inherently does not need this feature, and cannot make any use of it.

having said all of that, we _already_ support a way on agon-vdp to allow alternate software to run on the ESP32.  using the new "updater API" one can upload different firmware to the ESP32 and switch to that, which _can_ allow the user to run FabGL or ESP32Tiny apps.

we therefore already provide everything that is needed _within the Agon platform_ for this feature request.

it should be noted that at this point you are no longer running agon-vdp, and no longer have a machine running the complete Agon software platform.

what is _actually_ needed in order to facilitate running FabGL or ESP32Tiny programs on Agon hardware is two components, neither of which really have a place as part of the Agon software platform.

the first feature is software running on the eZ80 to accept requests from the ESP32 for SD Card IO operations, and to send back responses.  this can be built on top of existing MOS APIs, so there is no Agon platform work for us to do here to facilitate this.

the second feature is library shims to add in to the source code of FabGL and/or ESP32Tiny apps to replace their SD card access with serial messages thru to the aforementioned software running on the eZ80.  as these are not applications that run on the Agon platform itself, this work is outside the scope of the agon-vdp project.

snakebyte69, 2023-11-17:
"the idea of adding features to agon-vdp in order to support running FabGL or ESP32Tiny apps presumes the ability to run those apps within the current VDP software platform. agon-vdp simply is not architected to allow that to be possible."

Agon-VDP is not involved at all and not needed, this is a FabGL only proposal with MOS facilitating the access to the SDCard.
All of the FabGL apps already run on the Agon Light except for the parts that need the SDCard.
e.g. Space Invaders, Altair 8800, CP/M 3.
RunCPM is a good example of an app that runs but crashes when it needs to access the SDCard.
MOS will need to be modified and potentially FabGL.

"as I see it, this is therefore not a feature that belongs as part of the Agon software platform. software written for the Agon software platform inherently does not need this feature, and cannot make any use of it."

The Agon can be whatever anyone wants it to be where the code is written for it, it's not any one person's version of the paradigm.

oldpatientsea, 2023-11-17:
>> Agon-VDP is not involved at all and not needed, this is a FabGL only proposal 

Exactly. And this is not a FabGL board. Which is why it is out of scope.

>> The Agon can be whatever anyone wants it to be where the code is written for it, it's not any one person's version of the paradigm.

Of course. And you are free to write the code for it just as Steve and Dean are welcome to wish you luck but not give up even more of their free time to write it for you.

stevesims, 2023-11-17:
> Agon-VDP is not involved at all and not needed, this is a FabGL only proposal with MOS facilitating the access to the SDCard. All of the FabGL apps already run on the Agon Light except for the parts [...]

as I said in my previous comment, MOS already provides the facilities needed.  a program running on top of MOS is all that is required for this feature.  no changes to MOS are necessary.

snakebyte69, 2023-11-18:
"MOS already provides the facilities needed. a program running on top of MOS is all that is required for this feature. no changes to MOS are necessary."

Very useful information!
If I understood it I would write it myself.