breakintoprogram/agon-vdp #102: Latest code does not run at all on some Agon HW when PSRAM is enabled.

lennart-benschop, 2023-09-29:
I have an original Agon Light from the first series.

I updated MOS (and Basic to the very latest versions).

Until mid August I could always compile the latest Agon VDP software using the Arduino IDE under Linux and upload it to my board using the recommended settings, including "PSRAM Enabled". This worked.

When I compile the latest version with "PSRAM Enabled", it compiles without warnings, it uploads to Agon without any issues, but Agon does not start up. No sound or video whatsoever.

I will try to narrow it down to the exact git commit where the problem starts to occur.

When I keep everything the same and just set "PSRAM DIsabled", everything works when I upload the compiled VDP code.
In the past I did run software that absolutely required PSRAM and this worked.

It's absolutely the hardware at fault, not the Arduino setup:
- I tried to upload the precompiled binary (posted Sep 28, 2023 by Jeroen Venema), both on Windows and on Linux. In both
  cases the upload procedure appears to complete without errors, my board just does not work.
- I tried to compile on both IDE 2.2.1 and 1.8.9 under Linux with the correct ESP toolchain v2.0.4 and the correct versions of ESP32Time and vdp-gl. Same result with PSRAM Enabled, runs with PSRAM Disabled. Works with PSRAM Enabled when compiling an older version of the VDP code.
-

lennart-benschop, 2023-09-29:
Seems to break on the following commit:

9d9bd7cd6c34497fb23019a3dc39fc0a733f340f
Author: Steve Sims <steve@stevesims.dev>
Date:   Wed Aug 9 15:24:56 2023 +0100

    improve audio memory management
    
    use C++ vectors and smart pointers

lennart-benschop, 2023-09-29:
But before just condemning the hardware: never rule out an uninitialized variable bug. I guess modern gcc finds most on its own if enough warnings are enabled, but a good static code check can be worth your trouble.

lennart-benschop, 2023-10-04:
When I enable the serial monitor in the Arduino ToolKit and #define DEBUG 1 in video.ino, I get the following output:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
wait_eZ80: Start

wait_eZ80: End

init_audio: we have reserved 32 channels

audio_driver: setWaveform 0

audio_driver: setWaveform 0 done

audio_driver: init 0

free mem: 240424

audio_driver: setWaveform 0

audio_driver: setWaveform 0

audio_driver: setWaveform 0 done

audio_driver: init 2

free mem: 122628

audio_driver: setWaveform 0 done

audio_driver: init 1

free mem: 122628

after change of canvas...

  free internal: 166084

  free 8bit: 122628

  free 32bit: 166084

do_modeChange: canvas(640,480), scale(2.000000,2.133333), mode 1, videoMode 0

audio_driver: play_note 0,100,750,125

audio_driver: play 0,100,750,125

audio_driver: elapsed 126

audio_driver: end

----
When I run with PSRAM enbaled, output stops at "Wait_eZ80: Start". It's the only thing printed from the VDP firmware, the previous output is from the internal boot loader, which I also get when not enabling DEBUG

lennart-benschop, 2023-10-04:
Probably the UART RX buffer gets allocated in PSRAM and as PSRAM is not reliable on my device, the wrong bytes get returned from it. Something like this must be it.

lennart-benschop, 2023-10-18:
I re-flashed MOS with the official latest 1.04RC2 version and now as of a sudden, VDP with PSRAM enabled works again. When I first discovered the problem I did make sure to update MOS to the then latest version. So the original Agon flies again.

closed 2023-10-18 by lennart-benschop

stevesims, 2023-10-22:
yup - for a while there was a version of MOS that didn't wait long enough for the VDP to start up - oops :grin: