breakintoprogram/agon-vdp #113: vdu 25,80,x;y; do wrong triangles

DjPoke, 2023-10-18:
Hi,
vdu 25,80,x;y; do wrong triangles with firmware 1.0.4 RC1+.

I'm trying to draw filled and not filled rectangles in asm.
fn_rect function draw a not filled rectangle, but if I draw a filled rectangled (based on two triangles) at top of it, the first rectangle is not fully removed.

```
; draw a rectangle
fn_rect:
	vdu 25
	vdu 4
	ld ix,x1
	ld a,(ix + 0)
	vdu_a
	ld a,(ix + 1)
	vdu_a
	ld iy,y1
	ld a,(iy + 0)
	vdu_a
	ld a,(iy + 1)
	vdu_a

	vdu 25
	vdu 5
	ld ix,x2
	ld a,(ix + 0)
	vdu_a
	ld a,(ix + 1)
	vdu_a
	ld iy,y1
	ld a,(iy + 0)
	vdu_a
	ld a,(iy + 1)
	vdu_a

	vdu 25
	vdu 5
	ld ix,x2
	ld a,(ix + 0)
	vdu_a
	ld a,(ix + 1)
	vdu_a
	ld iy,y2
	ld a,(iy + 0)
	vdu_a
	ld a,(iy + 1)
	vdu_a

	vdu 25
	vdu 5
	ld ix,x1
	ld a,(ix + 0)
	vdu_a
	ld a,(ix + 1)
	vdu_a
	ld iy,y2
	ld a,(iy + 0)
	vdu_a
	ld a,(iy + 1)
	vdu_a

	vdu 25
	vdu 5
	ld ix,x1
	ld a,(ix + 0)
	vdu_a
	ld a,(ix + 1)
	vdu_a
	ld iy,y1
	ld a,(iy + 0)
	vdu_a
	ld a,(iy + 1)
	vdu_a
	
	ret

; draw a filled rectangle
fn_rectf:
	ld ix,x1
	ld iy,y1
	
	vdu 25
	vdu 4
	ld a,(ix+0)
	vdu_a
	ld a,(ix+1)
	vdu_a
	ld a,(iy+0)
	vdu_a
	ld a,(iy+1)
	vdu_a
	
	ld ix,x2
	ld iy,y1

	vdu 25
	vdu 4
	ld a,(ix+0)
	vdu_a
	ld a,(ix+1)
	vdu_a
	ld a,(iy+0)
	vdu_a
	ld a,(iy+1)
	vdu_a
		
	ld ix,x2
	ld iy,y2
	
	vdu 25
	vdu 80
	ld a,(ix+0)
	vdu_a
	ld a,(ix+1)
	vdu_a
	ld a,(iy+0)
	vdu_a
	ld a,(iy+1)
	vdu_a
	
	ld ix,x1
	ld iy,y1
	
	vdu 25
	vdu 4
	ld a,(ix+0)
	vdu_a
	ld a,(ix+1)
	vdu_a
	ld a,(iy+0)
	vdu_a
	ld a,(iy+1)
	vdu_a
		
	ld ix,x1
	ld iy,y2
	
	vdu 25
	vdu 4
	ld a,(ix+0)
	vdu_a
	ld a,(ix+1)
	vdu_a
	ld a,(iy+0)
	vdu_a
	ld a,(iy+1)
	vdu_a
		
	ld ix,x2
	ld iy,y2
	
	vdu 25
	vdu 80
	ld a,(ix+0)
	vdu_a
	ld a,(ix+1)
	vdu_a
	ld a,(iy+0)
	vdu_a
	ld a,(iy+1)
	vdu_a

```

breakintoprogram, 2023-10-18:
@DjPoke have you tried 1.04 RC2? That's got more support for the VDU plot commands than RC1.

DjPoke, 2023-10-18:
> @DjPoke have you tried 1.04 RC2? That's got more support for the VDU plot commands than RC1.

Under RC2, with my real AgonLight2, triangles are almost entirely black.

DjPoke, 2023-10-18:
This is a photography of my Agon Light 2 not drawing triangles.
[screenshot](https://cpclike.com/wp-content/uploads/2023/10/realcap-scaled.jpg)

lennart-benschop, 2023-10-18:
1.04 RC2 honours the plot mode, therefore 80 is a relative move (without drawing anything). For a triangle you should now use PLOT 85 instead of PLOT 80.

DjPoke, 2023-10-18:
Thanks for the information. Just checked on RC2, there is no more the black color issue.
Now, RC2 do like RC1 : a triangle is not drawn exactly at the same place than a line. There is an offset of 1 pixel.

DjPoke, 2023-10-19:
If it can help: drawing a line from x1 to x2 is equal to drawing a line from x1+width-1.
I think the bug is something like this, just one pixel of difference.

stevesims, 2023-10-22:
there is every chance that there's an off-by-one issue with filled shapes - the underlying graphics system has a concept of "line" and "fill", which doesn't quite fit with what we're doing - so we _might_ need to tweak things slightly

FYI there is now a "rectangle fill" plot command range - they're numbered 0x60-67, so the "absolute plot in current foreground colour" version of that is 0x65 (or 101 in decimal)

closed 2023-11-06 by breakintoprogram