breakintoprogram/agon-vdp #137: Circle terminal position is incorrect.

paulscottrobson, 2023-11-11:
When drawing a circle using the GXR the position of the cursor afterwards is the upper right point of the bounding box.  On a BBC Master 128 (tested on JSBEEB) it is on the circumference of the circle at 45 degrees e,g, at offset cos(45).r sin(45).r from the circle centre .... probably because it draws the circle using reflection / Bresenham and this is where it ends up.

Now, this is arguably very dozy behaviour. I'd have implemented it so it drew an ellipse in a bounding box (not fun on a 65C02) or so it reset the position to the centre of the circle, useful for drawing rings with outlines.

I would suggest rather than copying this odd feature, which I doubt any developer has ever used, the documentation says that the position of the graphics cursor after drawing the circle is indeterminate.

stevesims, 2023-11-11:
there _is_ a bug here in the Agon plot implementation, but it's not really what you're describing.

Acorn's VDU system has the concept of a "graphics cursor", but essentially, simplified, it just keeps track of the last point given in a drawing command.

to draw a circle you first need to move to a point at x,y coordinates which will be the centre of the circle, and you then need to use a circle plot command to a different x, y position.  when the circle is draw, the graphics cursor should be left at that x,y position.

that is the behaviour you have been seeing in JSBeeb.  the cursor is left where you told it to be positioned, i.e. the coordinates you gave to the circle PLOT command.  that can actually be anywhere around the circumference of the circle, not just at a position of 45 degrees.  rather than being "very dozy" I'd argue this behaviour is highly logical and easily understood.

the bug in the Agon plot implementation is that it is not updating the graphics cursor position after drawing the circle, so is leaving the graphics cursor at the centre of the circle.  (in both cases the graphics cursor position is completely determinate.). this likely affects several other plot operations too.  I will raise a pull request to fix this.

closed 2023-11-13 by breakintoprogram

paulscottrobson, 2023-11-15:
Yes ... but it doesn't. That was the point. On the BBC Master, which has the GXR built in, it leaves it on the circle circumference. This is *probably* because they do an algorithm for 45 degrees on the circle ; you can do the rest with mirroring ; flipping ; rotating, and the line 1/2 way up to the top was the last horizontal filled line drawn and they just forgot to put it back. The design of circle is really illogical. Rectangle drawing works as one might expect , providing the coordinates of opposite corners, whereas for circle you provide the radius, which really should be a bounding box, which could be limited to circles and is extensible to ellipses.

stevesims, 2023-11-15:
I ran up JSBeeb, selected BBC Master, drew some circles using the GXR command, and the cursor was always, consistently, left at the X,Y position used to plot the circle.  every single time.  it is categorically _not_ restricted to be at a 45 degree angle

as an example, try:
```
MOVE 500,500
PLOT &95,700,500
PLOT 5,200,200
```
you will see a circle drawn, centred on 500,500, and then a line from the right hand edge of the circle, starting at 700,500, down to 200,200.  that 700,500 point is most definitely not at 45 degrees to the centre of the circle - it's on the same line.  change that 700,500 to _anything you like_ and the circle will be drawn to intersect with that point, and the cursor left at the point given.

your interpretation of Acorn's circle drawing algorithm is simply incorrect.

their algorithm is for the user to specify the centre point, and then specify any point on the circumference of the circle.  I'd say that's pretty logical, and easily understood.  we're not going to adopt a different algorithm for the Agon.

the bug in the agon-vdp implementation, which has since been fixed, was that the Agon VDP wasn't moving the cursor after drawing the circle.  the cursor was being left and the centre.  when I fixed the bug I did so directly comparing the Agon VDP behaviour to the BBC Master GXR running on JSBeeb.  the behaviour how matches.  as an extra layer of verification, I also ran up an Archimedes emulator, and behaviour matches there too

Acorn's VDU system _does_ also support drawing ellipses but that is _not_ done using a "bounding box" approach.  if I've understood what you're suggesting, that would not provide true ellipse support, but rather squashed circles, which are a different thing and not directly supported by Acorn's VDU system.  for now we don't offer any ellipse support.  if this ever gets added, we are more likely to support Acorn's style of ellipses than the simpler squashed circles

PR #138 (2023-11-13): update graphics cursor correctly
plus slightly optimise our use of the graphics plot operation queue for a small graphics performance improvement

fixes #137