breakintoprogram/agon-vdp #142: When using double-buffered modes, keyboard input is not visible

breakintoprogram, 2023-11-11:
To replicate

Switch to MODE 132
Try typing something in

stevesims, 2023-11-11:
yep - but this is just the nature of the double-buffered screen modes

all drawing in double-buffered modes happens in the back buffer, so inevitably that will include relaying out to screen keyboard input

rendering to the _front_ buffer would negate the point of having double-buffered modes :grin:

you can see what stuff you've input by typing `VDU 23,0,195` to swap buffers

whilst this behaviour may, at first glance, be unexpected, it is correct

one could try to come up with contrived ways of working around this in order to allow the CLI to operate in these modes, but I don't really see that as something that belongs in the VDP codebase.  the "fix" would likely involve some convoluted code that would need to swap buffers whilst keyboard input was happening to ensure that keystrokes appear on the front buffer.  this would likely at best be pretty problematic

I think the solution here is to just better document the nature of double-buffered modes

julianregel, 2023-11-11:
Nothing really significant to add to Steve's comment, but I'd class this as "Working as intended".

The intention is to allow the programmer to draw everything to the back buffer and then switch when they have done so at a point of their choosing. Doing per-character updates would be very difficult.

breakintoprogram, 2023-11-12:
That's fine - just flagging it up with the hive mind so that if it does come up, I can say we've discussed it.

closed 2023-11-12 by breakintoprogram