breakintoprogram/agon-vdp #145: Feature request: expose the selected FG/BG colours

avalonbits, 2023-12-07:
Today it is not possible to know which FG/BG colors are being used without resorting to some printing a character on the screen, and then querying for the pixel colors at different locations of  the character ([this is what AED does today](https://github.com/avalonbits/aed/blob/99ec534cc749b44f8c55a4527ca46591f751bda2/src/screen.c#L92)).

the FG/BG colour should be part of the mode information sent as is width/height/color count.

avalonbits, 2023-12-07:
I like the idea of a separate command, is this is typically not required info. AED needs it so it can correctly display the cursor without blinking and on exit it can restore what was the color scheme being used if it were changed in the editor i.e being a "good citizen".

stevesims, 2023-12-15:
I've just come up with an implementation of this in AgonConsole8/agon-vdp#112

@avalonbits if you could take a look at that, assuming you think it's good I'll raise an equivalent PR here too

avalonbits, 2023-12-15:
That looks nice! Very clear API.

stevesims, 2023-12-16:
PR now raised here too :grin: