breakintoprogram/agon-vdp #157: Agon locks up when POS and/or VPOS is used with VDU 31

tonedef71, 2023-12-23:
Any of the following lines of code in BASIC or BASIC ADL will cause the Agon to lock-up:

``` 
10 VDU 31, POS, VPOS
20 VDU 31, 0, VPOS
30 VDU 31, POS, 0
```

The VDP should handle the evaluation of `POS` and `VPOS` properly so that the proper X and Y coordinates are passed to `VDU 31` .

lennart-benschop, 2023-12-23:
Problem does not occur with PRINT TAB(POS,VPOS); Neither when you first assign variables with POS & VPOS and pass the variables to VDU 31.

Probably the root cause: the BASIC VDU command passes each byte to the VDP after it is evaluated, so 31 gets passed immediately after it is evaluated. Then POS and/or VPOS are evaluated and BASIC sends the VDU 23, 0,130 sequence to read the cursor position and wait for the result. But the first bytes of this sequence get interpreted as the parameters of VDU 31, so the cursor read command will never finish. 

A fix would probably need to happen in BASIC, for example after each command that might output to the screen, call VDU 23,0,130 and store the cursor position locally. Evaluation of POS & VPOS just take those locally stored bytes. Or the VDU command has to buffer all bytes of that command into a local buffer and then output that buffer at once. The VDU command could buffer some of the VDU sequences until they are complete (VDU 23, 24, 28 and 31) and only output them as a whole.

Other versions of Beeb Basic do not send bytes to the VDU to retrieve cursor position information, so the problem does not occur here.

tonedef71, 2023-12-23:
> Problem does not occur with PRINT TAB(POS,VPOS); Neither when you first assign variables with POS & VPOS and pass the variables to VDU 31.
>
> Probably the root cause: the BASIC VDU command passes each [...]

@lennart-benschop Thank you for performing an analysis of the issue.  Does this issue now need to be opened in each of the Agon BASIC git repositories, https://github.com/breakintoprogram/agon-bbc-basic-adl/issues and https://github.com/breakintoprogram/agon-bbc-basic/issues?